borglab/gtsam #2104: Python docstring edge case

p-zach, 2025-04-19:
# The issue

The C++ documentation does not get ported into the Python docstring when the parameter names differ between the function definition in the class's header file and the function definition in the `.i` file. This is because `xml_parser.py` relies on parameter *names* to match documentation from the XML to a function when the function is overloaded; it cannot use parameter *types* because these are not guaranteed to be the same between the GTSAM implementation and the `.i` file which 'requests' the documentation  (e.g. `OptionalJacobian<6, 6>` in GTSAM becomes `Eigen::Ref<Eigen::MatrixXd>` in the `.i`).

# Concrete example

For reference, for the function
```c++
/// Log map at identity - return the canonical coordinates \f$ [R_x,R_y,R_z,T_x,T_y,T_z] \f$ of this rotation
static Vector6 Logmap(const Pose3& pose, OptionalJacobian<6, 6> Hpose = {});
```
the Doxygen XML entry looks like this:
```xml
<memberdef kind="function" id="a03936_1a544c6f78890801a1b2a10ac5ae24b589" prot="public" static="yes" const="no" explicit="no" inline="no" virt="non-virtual">
        <type>Vector6</type>
        <definition>Vector6 gtsam::Pose3::Logmap</definition>
        <argsstring>(const Pose3 &amp;pose, OptionalJacobian&lt; 6, 6 &gt; Hpose={})</argsstring>
        <name>Logmap</name>
        <qualifiedname>gtsam::Pose3::Logmap</qualifiedname>
        <param>
          <type>const <ref refid="a03936" kindref="compound">Pose3</ref> &amp;</type>
          <declname>pose</declname>
        </param>
        <param>
          <type><ref refid="a03128" kindref="compound">OptionalJacobian</ref>&lt; 6, 6 &gt;</type>
          <declname>Hpose</declname>
          <defval>{}</defval>
        </param>
        <briefdescription>
<para>Log map at identity - return the canonical coordinates <formula id="6">$ [R_x,R_y,R_z,T_x,T_y,T_z] $</formula> of this rotation. </para>
        </briefdescription>
        <detaileddescription>
        </detaileddescription>
        <inbodydescription>
        </inbodydescription>
        <location file="gtsam/geometry/Pose3.h" line="150" column="18" bodyfile="gtsam/geometry/Pose3.cpp" bodystart="258" bodyend="278"/>
```
And the corresponding `.i` entries look like this:
```c++
static gtsam::Vector Logmap(const gtsam::Pose3& p);
static gtsam::Vector Logmap(const gtsam::Pose3& p, Eigen::Ref<Eigen::MatrixXd> H);
```
Ordinarily in a case like this, the parser recognizes that the second argument is optional and can assign the docstring "Log map at identity ... rotation" to both of `Logmap(p)` and `Logmap(p, H)`. However, in this case, the parameter names do not match, so it does not recognize that XML as having documentation for either of the `Logmap`s. Of course, if the `.i` entries were changed to have parameter names `pose` and `pose, Hpose`, then they would match the XML and everything would work as expected.

# Possible resolutions

Currently, because of the above, parameter names listed in the `.i` file must match exactly the parameter names in the `.h` file, or the documentation will not be pulled through to the Python docstring. There are two possibilities to reduce the severity of this issue, but there is not to my knowledge a real fix.
1. Moving forward, expect that new functions added to `.i` files should match header file parameter names exactly, and fix existing occurrences in the `.i` files where the parameter names don't match the header. 
	- Benefit: fail-safe. If documentation is not found, it leaves the docstring blank. There is no chance of incorrect documentation unless a developer changes the `.i` parameter name(s) to exactly the name(s) of the parameters of a different overload of the same function.
	- Drawback: maintenance. Requires aware developers to keep parameter names matching and fix mismatches if missing documentation is encountered.
2. Alter the docstring generation code to ignore parameter names altogether and only use parameter count to determine which documentation to extract. 
	- Benefit: reduced maintenance. Documentation converts to docstring without having to worry about matching parameter names.
	- Drawback: fail-unsafe. There is a possibility that incorrect documentation could be assigned to a function, if the function has two (or more) overloads in the `.i` which take the same number of parameters, and the associated functions in the `.h` are not declared in the exact same order. See the following example.

I suppose a third option to fix the issue would be to work out a more reliable way to associate `.i` functions with their XML documentation. But I have found no more reliable way to map from `.i` function to XML function given the limited amount of information present in the `.i` declarations. I would welcome ideas on that front.

## Example for (2)

The Python documentation for `Pose3.transformTo` looks like this:
```
>>> help(gtsam.Pose3.transformTo)
Help on instancemethod in module gtsam.gtsam:

transformTo(...)
    transformTo(*args, **kwargs)
    Overloaded function.

    1. transformTo(self: gtsam.gtsam.Pose3, point: numpy.ndarray[numpy.float64[3, 1]]) -> numpy.ndarray[numpy.float64[3, 1]]

    takes point in world coordinates and transforms it to Pose coordinates
    point: point in world coordinates
    Returns: point in Pose coordinates

    2. transformTo(self: gtsam.gtsam.Pose3, point: numpy.ndarray[numpy.float64[3, 1]], Hself: numpy.ndarray[numpy.float64[m, n], flags.writeable, flags.f_contiguous], Hpoint: numpy.ndarray[numpy.float64[m, n], flags.writeable, flags.f_contiguous]) -> numpy.ndarray[numpy.float64[3, 1]]

    takes point in world coordinates and transforms it to Pose coordinates
    point: point in world coordinates
    Hself: optional 3*6 Jacobian wrpt this pose
    Hpoint: optional 3*3 Jacobian wrpt point
    Returns: point in Pose coordinates

    3. transformTo(self: gtsam.gtsam.Pose3, points: numpy.ndarray[numpy.float64[m, n]]) -> numpy.ndarray[numpy.float64[m, n]]

    transform many points in world coordinates and transform to Pose.
    points: 3*N matrix in world coordinates
    Returns: points in Pose coordinates, as 3*N Matrix
```

If we implemented (2) and, in `Pose3.h`, the function header for implementation (1) came *after* the function header for implementation (3), the documentation would look like this:

```
>>> help(gtsam.Pose3.transformTo)
Help on instancemethod in module gtsam.gtsam:

transformTo(...)
    transformTo(*args, **kwargs)
    Overloaded function.

    1. transformTo(self: gtsam.gtsam.Pose3, point: numpy.ndarray[numpy.float64[3, 1]]) -> numpy.ndarray[numpy.float64[3, 1]]

    transform many points in world coordinates and transform to Pose.
    points: 3*N matrix in world coordinates
    Returns: points in Pose coordinates, as 3*N Matrix

    2. transformTo(self: gtsam.gtsam.Pose3, point: numpy.ndarray[numpy.float64[3, 1]], Hself: numpy.ndarray[numpy.float64[m, n], flags.writeable, flags.f_contiguous], Hpoint: numpy.ndarray[numpy.float64[m, n], flags.writeable, flags.f_contiguous]) -> numpy.ndarray[numpy.float64[3, 1]]

    takes point in world coordinates and transforms it to Pose coordinates
    point: point in world coordinates
    Hself: optional 3*6 Jacobian wrpt this pose
    Hpoint: optional 3*3 Jacobian wrpt point
    Returns: point in Pose coordinates

    3. transformTo(self: gtsam.gtsam.Pose3, points: numpy.ndarray[numpy.float64[m, n]]) -> numpy.ndarray[numpy.float64[m, n]]

    takes point in world coordinates and transforms it to Pose coordinates
    point: point in world coordinates
    Returns: point in Pose coordinates
```

It is subtly wrong in this case: the docs for (1) and (3) are flipped. In other cases, it could be possible that the documentation would be flagrantly wrong.

# TLDR

TL;DR: We need to decide whether we want to do a little more work and guarantee that docs are correct if present (approach 1), or do a little less work and risk that docs might be wrong in rare cases (approach 2).

dellaert, 2025-04-19:
I think a likely reason why there is a mismatch between .h and .i is that the .h was updated to be more clear in subsequent commits. So, my proposed solution would be:
- fix all .i files
- document and insist the naming is the same for new wrapping

ProfFan, 2025-04-20:
The actual reason where a C++ type needs to be specified is to provide MATLAB compatibility. Pybind infers the argument types automatically. The MATLAB wrapper is much less capable in its abilities and is pretty brittle - we have no CI for it also.

ProfFan, 2025-04-20:
`Eigen::Ref` from my memories is a workaround for some issues with MATLAB wrapper. One has to dig into the original PRs to understand what was going on then...

dellaert, 2025-04-20:
I think my suggestion resolves this, right?
Porter, Let’s discuss the remaining issues on Thursday. I’ll close this issue.
Fan, more discussion happened on slack, I think we have this…

closed 2025-04-20 by dellaert