borglab/gtsam #2125: unittests defined in testHybridGuassianISAM fail

aminoh, 2025-05-03:
## Description

execution of `make check` on an UBUNTU 24.04 LTS fail on the develop branch. The error message is:
```
In file included from /usr/include/c++/13/string:51,
                 from /usr/include/c++/13/bits/locale_classes.h:40,
                 from /usr/include/c++/13/bits/ios_base.h:41,
                 from /usr/include/c++/13/streambuf:43,
                 from /usr/include/c++/13/bits/streambuf_iterator.h:35,
                 from /usr/include/c++/13/iterator:66,
                 from /usr/include/boost/concept_check.hpp:22,
                 from /home/amino/Tools/gtsam/gtsam/base/concepts.h:14,
                 from /home/amino/Tools/gtsam/gtsam/base/Testable.h:36,
                 from /home/amino/Tools/gtsam/gtsam/discrete/AlgebraicDecisionTree.h:21,
                 from /home/amino/Tools/gtsam/gtsam/discrete/DecisionTreeFactor.h:21,
                 from /home/amino/Tools/gtsam/gtsam/discrete/DiscreteConditional.h:21,
                 from /home/amino/Tools/gtsam/gtsam/discrete/DiscreteBayesNet.h:21,
                 from /home/amino/Tools/gtsam/gtsam/hybrid/tests/testHybridGaussianISAM.cpp:19:
In function ‘constexpr int std::__memcmp(const _Tp*, const _Up*, size_t) [with _Tp = long unsigned int; _Up = long unsigned int]’,
    inlined from ‘static bool std::__equal<true>::equal(const _Tp*, const _Tp*, const _Tp*) [with _Tp = long unsigned int]’ at /usr/include/c++/13/bits/stl_algobase.h:1185:27,
    inlined from ‘bool std::__equal_aux1(_II1, _II1, _II2) [with _II1 = const long unsigned int*; _II2 = const long unsigned int*]’ at /usr/include/c++/13/bits/stl_algobase.h:1219:43,
    inlined from ‘bool std::__equal_aux(_II1, _II1, _II2) [with _II1 = __gnu_cxx::__normal_iterator<const long unsigned int*, vector<long unsigned int, tbb::detail::d1::tbb_allocator<long unsigned int> > >; _II2 = __gnu_cxx::__normal_iterator<const long unsigned int*, vector<long unsigned int, tbb::detail::d1::tbb_allocator<long unsigned int> > >]’ at /usr/include/c++/13/bits/stl_algobase.h:1227:31,
    inlined from ‘bool std::equal(_II1, _II1, _II2) [with _II1 = __gnu_cxx::__normal_iterator<const long unsigned int*, vector<long unsigned int, tbb::detail::d1::tbb_allocator<long unsigned int> > >; _II2 = __gnu_cxx::__normal_iterator<const long unsigned int*, vector<long unsigned int, tbb::detail::d1::tbb_allocator<long unsigned int> > >]’ at /usr/include/c++/13/bits/stl_algobase.h:1557:30,
    inlined from ‘bool gtsam::Conditional<FACTOR, DERIVEDCONDITIONAL>::ConstFactorRangeIterator::operator==(const OTHER&) const [with OTHER = std::vector<long unsigned int, tbb::detail::d1::tbb_allocator<long unsigned int> >; FACTOR = gtsam::HybridFactor; DERIVEDCONDITIONAL = gtsam::HybridConditional]’ at /home/amino/Tools/gtsam/gtsam/inference/Conditional.h:91:26,
    inlined from ‘virtual void HybridGaussianISAMIncrementalEliminationTest::run(TestResult&)’ at /home/amino/Tools/gtsam/gtsam/hybrid/tests/testHybridGaussianISAM.cpp:71:3:
/usr/include/c++/13/bits/stl_algobase.h:108:32: error: argument 2 null where non-null expected [-Werror=nonnull]
  108 |         return __builtin_memcmp(__first1, __first2, sizeof(_Tp) * __num);
      |                ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13/bits/stl_algobase.h:108:32: note: in a call to built-in function ‘int __builtin_memcmp(const void*, const void*, long unsigned int)’
In function ‘constexpr int std::__memcmp(const _Tp*, const _Up*, size_t) [with _Tp = long unsigned int; _Up = long unsigned int]’,
    inlined from ‘static bool std::__equal<true>::equal(const _Tp*, const _Tp*, const _Tp*) [with _Tp = long unsigned int]’ at /usr/include/c++/13/bits/stl_algobase.h:1185:27,
    inlined from ‘bool std::__equal_aux1(_II1, _II1, _II2) [with _II1 = const long unsigned int*; _II2 = const long unsigned int*]’ at /usr/include/c++/13/bits/stl_algobase.h:1219:43,
    inlined from ‘bool std::__equal_aux(_II1, _II1, _II2) [with _II1 = __gnu_cxx::__normal_iterator<const long unsigned int*, vector<long unsigned int, tbb::detail::d1::tbb_allocator<long unsigned int> > >; _II2 = __gnu_cxx::__normal_iterator<const long unsigned int*, vector<long unsigned int, tbb::detail::d1::tbb_allocator<long unsigned int> > >]’ at /usr/include/c++/13/bits/stl_algobase.h:1227:31,
    inlined from ‘bool std::equal(_II1, _II1, _II2) [with _II1 = __gnu_cxx::__normal_iterator<const long unsigned int*, vector<long unsigned int, tbb::detail::d1::tbb_allocator<long unsigned int> > >; _II2 = __gnu_cxx::__normal_iterator<const long unsigned int*, vector<long unsigned int, tbb::detail::d1::tbb_allocator<long unsigned int> > >]’ at /usr/include/c++/13/bits/stl_algobase.h:1557:30,
    inlined from ‘bool gtsam::Conditional<FACTOR, DERIVEDCONDITIONAL>::ConstFactorRangeIterator::operator==(const OTHER&) const [with OTHER = std::vector<long unsigned int, tbb::detail::d1::tbb_allocator<long unsigned int> >; FACTOR = gtsam::HybridFactor; DERIVEDCONDITIONAL = gtsam::HybridConditional]’ at /home/amino/Tools/gtsam/gtsam/inference/Conditional.h:91:26,
    inlined from ‘virtual void HybridGaussianISAMIncrementalEliminationTest::run(TestResult&)’ at /home/amino/Tools/gtsam/gtsam/hybrid/tests/testHybridGaussianISAM.cpp:83:3:
/usr/include/c++/13/bits/stl_algobase.h:108:32: error: argument 2 null where non-null expected [-Werror=nonnull]
  108 |         return __builtin_memcmp(__first1, __first2, sizeof(_Tp) * __num);
      |                ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13/bits/stl_algobase.h:108:32: note: in a call to built-in function ‘int __builtin_memcmp(const void*, const void*, long unsigned int)’
cc1plus: all warnings being treated as errors
make[3]: *** [gtsam/hybrid/tests/CMakeFiles/testHybridGaussianISAM.dir/build.make:76: gtsam/hybrid/tests/CMakeFiles/testHybridGaussianISAM.dir/testHybridGaussianISAM.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:12296: gtsam/hybrid/tests/CMakeFiles/testHybridGaussianISAM.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2598: CMakeFiles/check.dir/rule] Error 2
make: *** [Makefile:199: check] Error 2
```

## Steps to reproduce

1. attempt a fresh installation

## Expected behavior

N/A
## Environment

Ubuntu 24.04 LTS. cmake configuration is as follows:
```
- The CXX compiler identification is GNU 13.3.0
-- The C compiler identification is GNU 13.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
[... 127 lines cut ...]
    inlined from ‘Derived& Eigen::MatrixBase<Derived>::operator=(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<double, -1, -1>; Derived = Eigen::Block<Eigen::Matrix<double, -1, -1>, -1, -1, false>]’ at /home/amino/Tools/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/Assign.h:66:28,
    inlined from ‘void gtsam::JacobianFactor::fillTerms(const TERMS&, const gtsam::Vector&, const gtsam::SharedDiagonal&) [with TERMS = std::map<long unsigned int, Eigen::Matrix<double, 1, 1, 0, 1, 1> >]’ at /home/amino/Tools/gtsam/gtsam/linear/JacobianFactor-inl.h:90:14:
/usr/lib/gcc/x86_64-linux-gnu/13/include/emmintrin.h:169:19: warning: ‘void* __builtin_memcpy(void*, const void*, long unsigned int)’ reading 16 or more bytes from a region of size 8 [-Wstringop-overread]
  169 |   *(__m128d *)__P = __A;
      |   ~~~~~~~~~~~~~~~~^~~~~
In file included from /home/amino/Tools/gtsam/gtsam/3rdparty/Eigen/Eigen/Core:166:
In function ‘void* Eigen::internal::aligned_malloc(std::size_t)’,
    inlined from ‘void* Eigen::internal::conditional_aligned_malloc(std::size_t) [with bool Align = true]’ at /home/amino/Tools/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/util/Memory.h:241:24,
    inlined from ‘T* Eigen::internal::conditional_aligned_new_auto(std::size_t) [with T = double; bool Align = true]’ at /home/amino/Tools/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/util/Memory.h:404:69,
    inlined from ‘void Eigen::DenseStorage<T, -1, -1, -1, _Options>::resize(Eigen::Index, Eigen::Index, Eigen::Index) [with T = double; int _Options = 0]’ at /home/amino/Tools/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/DenseStorage.h:486:85,
[... cut ...] /stl_pair.h:586:22,
    inlined from ‘void gtsam::JacobianFactor::fillTerms(const TERMS&, const gtsam::Vector&, const gtsam::SharedDiagonal&) [with TERMS = std::map<long unsigned int, Eigen::Matrix<double, 1, 1, 0, 1, 1> >]’ at /home/amino/Tools/gtsam/gtsam/linear/JacobianFactor-inl.h:80:37:
/home/amino/Tools/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/util/Memory.h:182:20: note: source object of size 8 allocated by ‘malloc’
  182 |     result = malloc(size);
      |              ~~~~~~^~~~~~

```

ProfFan, 2025-05-04:
@varunagrawal

varunagrawal, 2025-05-04:
This seems to be a compiler or Eigen issue. Pytorch has a similar problem https://github.com/pytorch/pytorch/issues/99278#issuecomment-1787744264

Can you try using another Eigen version?

aminoh, 2025-05-11:
@varunagrawal Thank you for your input. I didn't get to try your suggestion precisely but in that same line, I built the targets with all the tests without any build errors using clang++ instead.

varunagrawal, 2025-05-11:
This is a @dellaert question. Frank wanted all warnings to come up as errors sometime early this year, but it seems like the error you're getting is more of a C++ standards one (23 versus our supported 17).

I recommend creating a new issue for that, and tagging Frank in it so he can assign someone appropriate to fix it. Or if you can wait until June, I can tackle it then.

dellaert, 2025-05-12:
Fix: include IncrementalFixedLagSmoother from gtsam, not gtsam_unstable

aminoh, 2025-05-12:
Alright, thanks very much both :) That hotfix removes the obstacle for me.
However I leave it up to you to close the issue or otherwise.

dellaert, 2025-05-12:
I’m unclear if the hotfix was applied to GTSAM code or your own code. I suspect the latter as otherwise our CI would not pass. But if the former, @varunagrawal, we should not close but fix.

aminoh, 2025-05-12:
@dellaert I fixed it yeah. It was an easy fix. I will create a pull request. Could you please re-open this issue?

varunagrawal, 2025-05-12:
@dellaert the issue is that @p-zach used `#warning` to indicate a compiler warning for `gtsam_unstable/IncrementalFixedLagSmoother` but `#warning` is a C++23 feature. Unless the user uses a suitable compiler, it will fail regardless. The only fix is to either use a `static_assert` to raise a compile time error, or ask users to use a compiler version with C++23 support.

p-zach, 2025-05-13:
@varunagrawal I see. I moved IncrementalFixedLagSmoother in the same way that [BatchFixedLagSmoother](https://github.com/borglab/gtsam/blob/develop/gtsam_unstable/nonlinear/BatchFixedLagSmoother.h) was moved. Maybe we should change the boilerplate for moved classes if the #warning isn't widely supported.

p-zach, 2025-05-13:
I know :) just wanted to convey that similar issues could arise elsewhere in the code so we know in the future