borglab/gtsam #2127: AprilTag/fiducial mapping using PlanerProjectionFactor

mcm001, 2025-05-07:
## Feature

We should consider modifying PlanerProjectionFactor to work with visual fiducials like AprilTags.

## Motivation

PlanerProjectionFactor was created in https://github.com/borglab/gtsam/pull/1934 , and provides a convenient way to deal with "body flat on ground at z=0" if you have a single fixed or free landmark. But I'm not sure I see a way to make it work with this problem setup:

- Have a robot flat on the ground at an unknown `Pose2`
- Robot2camera transform is an unknown `Pose3`
- My environment has an AprilTag or similar in it. The tag's origin is located at an unknown world2tag `Pose3`
  - Each of these tags has 4 corners, and the tag's size is known. The corner ordering is also consistent. 
  - The image below shows some example output, with tag corners shown in red and the tag's coordinate system shown in RGB for XYZ.
  [image]

The way I've implemented this right now is [using ExpressionFactors](https://github.com/mcm001/gtsam-playground/blob/ba069016120d39ebabd2d4325434be1edf5e6aa1/src/sfm_mapper/sfm_mapper.cpp#L191C1-L206C31):

```cpp
      gtsam::Key tagKey{helpers::TagIdToKey(tag.id)};
      std::vector<gtsam::Point3_> worldPcorners{TagModel::WorldToCornersFactor(tagKey)}; // 
      const gtsam::Pose3_ worldTbody_fac{GetNearestStateToKeyframe(keyframe.time)};

      constexpr int NUM_CORNERS = 4;
      for (size_t i = 0; i < NUM_CORNERS; i++) {
        // Decision variable - where our camera is in the world
        const gtsam::Pose3_ robotTcamera_wpilib{keyframe.cameraIdx};
        const gtsam::Pose3_ robotTcamera_cv{robotTcamera_wpilib *
                                     wpilibToCvCameraTransform};

        // Where we'd predict the i'th corner of the tag to be
        const gtsam::Point2_ prediction = PredictLandmarkImageLocationFactor(
            worldTbody_fac, robotTcamera_cv, cameraCalMap[keyframe.cameraIdx],
            worldPcorners[i]);

        // where we saw the tag's i'th corner in the image
        gtsam::Point2 measurement = {tag.corners[i].x, tag.corners[i].y};

        // Add this prediction/measurement pair to our graph
        graph.addExpressionFactor(prediction, measurement, cameraNoiseModel);
```

## Pitch

A couple options I see:
- Create a variant of PlanerProjectionFactor that allows a single static `Point3` offset from "landmark" origin to the thing we observed
- Modify PlanerProjectionFactor to allow up to N many landmarks per landmark ("landmark" is a poor name)

## Alternatives

The ExpressionFactor approach above -does- work. Nothing wrong with it :) -- but it might be slower? Not sure.

## Additional context

All my prototyping code's been living at https://github.com/mcm001/gtsam-playground/tree/camera-extrinsic-cal/src/sfm_mapper

dellaert, 2025-05-08:
I replied in GTSAM user group - we have code, but need volunteers to hep PR it into GTSAM :-)

mcm001, 2025-05-09:
Oh awesome! I did send a join request to the Google users group, looks like I'm just waiting for someone to let me in 😀 

I wasn't able to find the code you mentioned on GitHub (at least a cursory search came up empty) -- do you know where it ended up?

dellaert, 2025-05-09:
It’s not on github :-)

dellaert, 2026-08-16:
Closing as no response. Feel free to create a PR, I’ll review.

closed 2026-08-16 by dellaert