borglab/gtsam #2146: Some issues in hybrid

dellaert, 2025-05-26:
## Description

Two small issues I encountered while helping to document hybrid:
- choose in a HybrdiBayesTree yields Gaussian cliques that are empty, presumably from discrete only cliques
- prune in a HybridBayesNet with max_leaves == 1 should convert to entirely non-hybrid, I think

## Steps to reproduce

1. Try display in HybridBayesTree.ipynb 
2. Try removing marginal threshold in HybridBayesNet.ipynb

## Additional information

Notebooks are still in unmerged docs-hybrid branch at the moment

varunagrawal, 2025-06-13:
#2170 should have resolved part 1. For part 2, I think it should still be hybrid since 1 leaf does not mean we have made a choice on the discrete assignment. Another case is when more than one leaf in the DiscreteConditional has the same value so they get merged together. It is now 1 leaf but not a specific assignment.

`PrintDiscreteValues` works in Jupyter, but it prints out of order due to [different buffers for C++ and Python](https://pybind11.readthedocs.io/en/stable/advanced/pycpp/utilities.html).

varunagrawal, 2025-06-13:
I've fixed `PrintDiscreteValues` in #2120.