borglab/gtsam #2161: 4.3a0 Build issue with error: variable ‘m_runtime’ set but not used

PeterQFR, 2025-06-04:
## Description

Building tag v4.3a0 with the following changes to turn off Boost Features
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index cabde7653..ce461ce57 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -68,8 +68,8 @@ include(GtsamPrinting)
 
 ############### Decide on BOOST ######################################
 # Enable or disable serialization with GTSAM_ENABLE_BOOST_SERIALIZATION
-option(GTSAM_ENABLE_BOOST_SERIALIZATION "Enable Boost serialization" ON)
-option(GTSAM_USE_BOOST_FEATURES "Enable Features that use Boost" ON)
+option(GTSAM_ENABLE_BOOST_SERIALIZATION "Enable Boost serialization" OFF)
+option(GTSAM_USE_BOOST_FEATURES "Enable Features that use Boost" OFF)
 
 if(GTSAM_ENABLE_BOOST_SERIALIZATION OR GTSAM_USE_BOOST_FEATURES)
 include(cmake/HandleBoost.cmake)
```
in a 20.04 dockerfile and I get:
 
```

gtsam/gtsam/navigation/ManifoldEKF.h:162:11: error: variable ‘m_runtime’ set but not used [-Werror=unused-but-set-variable]
  162 |       int m_runtime;
      |           ^~~~~~~~~
cc1plus: all warnings being treated as errors
```

Not sure of the issue here. I'm building inside a ros2 galactic workspace with colcon, and I've not seen this error before.

ProfFan, 2025-06-04:
It's a warning, you can either locally say
```
(void)m_runtime;
```
instead of printf

or turn off that warning in your CXXFLAGS: `-Wno-nused-but-set-variable`

WZG3661, 2025-06-06:
Comment out the line 134 in cmake/GtsamBuildTypes.cmake

DanielChaseButterfield, 2025-06-10:
Commenting out line 134 in `cmake/GtsamBuildTypes.cmake` unfortunately didn't work for me, but adding `(void)m_runtime;` to line 182 of `gtsam/gtsam/navigation/ManifoldEKF.h` worked for me.

dellaert, 2025-06-10:
This is a bug. I think the runtime m should be defined only in the first case of the const expression if statement. 

Unfortunately, it doesn’t trigger our CI. Could you see whether making that change fixes the issue in your container and if so, possibly do a PR from your fork?

DanielChaseButterfield, 2025-06-11:
That change also worked for my container! I've opened a PR with the change, so assuming it passes test cases it should be good to go. Here's hoping it works for @PeterQFR as well.

closed 2025-06-12 by dellaert