borglab/gtsam #2179: `make python-install` will not install into virtual environment

CodeXTL, 2025-06-27:
## Description

Followed the directions given in `gtsam/python/README.md` to build the Python wrappers for gtsam. The command `make python-build` functions as expected. However, the directions specify that the same command can be used to install into a virtual environment if the virtual environment is active, but this did not work for me. Even when I ran the command with an active virtual environment, it installed the wrapper package into my base interpreter. 

## Steps to reproduce

1. Navigate to `gtsam` directory
2. Create new virtual environment: `python -m venv .env`
3. Activate new virtual environment: `source .env/bin/activate`
4. Navigate to `gtsam/build`
5. Run `cmake .. -DGTSAM_BUILD_PYTHON=1 -DGTSAM_PYTHON_VERSION=3.6.10`
6. Run `make python-install`
7. Check if installation successful: `pip list`

## Expected behavior
When `pip list` is ran with the virtual environment active, the list should include the package `gtsam` with version `4.3a0`.

## Environment
- Ubuntu Linux 24.04
- Virtual environment with Python 3.12.3
- Virtual environment with pip 25.1.1
- CMake version 3.28.3
- GNU Make 4.3

## Additional information

CodeXTL, 2025-06-27:
Honestly not something very urgent. I can personally take a look to see why this is the case, but someone more qualified might be able to fix it faster.

ProfFan, 2025-06-29:
Yeah we need to clean up the logic for Python, also #2182 

The following stuff needs to be split:
1. Interpreter and libs used to build bindings
2. Current interpreter that the user want to use
3. Maybe, remove `python-install` (which is really confusing) and make the wheel directly, so the user just use the built wheel instead

dellaert, 2025-06-30:
Hmmmm, I really like python-install, makes things easy

ProfFan, 2025-07-07:
It also creates some confusion which Python environment GTSAM in being installed into. I believe the current behavior is installing into the environment used to build GTSAM itself. We should document that...

talregev, 2025-08-28:
we do install on the ci and it working, also testing it on local.
can you try to do the install like that:
cmake --build . --target python-install