borglab/gtsam #2202: GTSAM 4.2.0 compiles successfully and builds fine in the catkin project, but it errors at runtime.

zhangqi-sudo, 2025-07-25:
if (NOT MSVC)
  option(GTSAM_BUILD_WITH_MARCH_NATIVE  "Enable/Disable building with all instructions supported by native architecture (binary may not be portable!)" OFF)

cmake \
  -DCMAKE_INSTALL_PREFIX=/usr/local \
  -DBUILD_SHARED_LIBS=ON \
  -DGTSAM_USE_SYSTEM_EIGEN=ON \
  -DEIGEN3_INCLUDE_DIR=/usr/include/eigen3 \
  -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF \
  -DCMAKE_BUILD_TYPE=Release \
  ..

I’m using GTSAM version 4.2.0 in my fast‑livo2 project, but I’m encountering the same error. I’ve confirmed that I’m using the system Eigen and have disabled -march=native in my CMake configuration, yet the error still occurs:

==26415== Invalid free() / delete / delete[] / realloc()
==26415==    at 0x483CA3F: free (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==26415==    by 0x68698D0: _GLOBAL__sub_I_lago.cpp (in /usr/local/lib/libgtsam.so.4.2.0)
==26415==    by 0x4011B99: call_init.part.0 (dl-init.c:72)
==26415==    by 0x4011CA0: call_init (dl-init.c:30)
==26415==    by 0x4011CA0: _dl_init (dl-init.c:119)
==26415==    by 0x4001139: ??? (in /usr/lib/x86_64-linux-gnu/ld-2.31.so)
==26415==    by 0x2: ???
==26415==    by 0x1FFEFFFCE2: ???
==26415==    by 0x1FFEFFFD24: ???
==26415==    by 0x1FFEFFFD39: ???
==26415==  Address 0x1905cda0 is 16 bytes inside a block of size 56 alloc'd
==26415==    at 0x483B7F3: malloc (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==26415==    by 0x12EDB8: Eigen::PlainObjectBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> >::PlainObjectBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >(Eigen::DenseBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> > const&) (in /home/zhangqi/slam/catkin_ws/devel/lib/fast_livo/fastlivo_mapping)
==26415==    by 0x68698B2: _GLOBAL__sub_I_lago.cpp (in /usr/local/lib/libgtsam.so.4.2.0)
==26415==    by 0x4011B99: call_init.part.0 (dl-init.c:72)
==26415==    by 0x4011CA0: call_init (dl-init.c:30)
==26415==    by 0x4011CA0: _dl_init (dl-init.c:119)
==26415==    by 0x4001139: ??? (in /usr/lib/x86_64-linux-gnu/ld-2.31.so)
==26415==    by 0x2: ???
==26415==    by 0x1FFEFFFCE2: ???
==26415==    by 0x1FFEFFFD24: ???
==26415==    by 0x1FFEFFFD39: ???
==26415== 
==26415== 
==26415== Exit program on first error (--exit-on-first-error=yes)

I would greatly appreciate any guidance from the experts—thank you so much!

ProfFan, 2025-07-26:
This seems to be the issue fixed in #2148 , please test with master or cherry-pick the commit

zhangqi-sudo, 2025-07-28:
> 这似乎是在[#2148](https://github.com/borglab/gtsam/pull/2148)中解决的问题，请测试master或 cherry-pick 提交

thanks！But why am I still encountering that error even though I disabled LAGO when compiling GTSAM?

zhangqi-sudo, 2025-07-28:
> This seems to be the issue fixed in [#2148](https://github.com/borglab/gtsam/pull/2148) , please test with master or cherry-pick the commit

Do I just need to download the source code from https://github.com/borglab/gtsam/tree/develop and compile the library

zhangqi-sudo, 2025-07-29:
> This seems to be the issue fixed in [#2148](https://github.com/borglab/gtsam/pull/2148) , please test with master or cherry-pick the commit

I compiled and tested the fix/unsafe_static_identity_matrices_in_lago version, and when running with -fsanitize=address, it still reports this error.

zhangqi-sudo, 2025-07-29:
> 这似乎是在[#2148](https://github.com/borglab/gtsam/pull/2148)中解决的问题，请测试master或 cherry-pick 提交

The issue is resolved: the alignment settings must be consistent. Because GTSAM was built with -march=native disabled, none of your project’s CMake files—including subprojects and any target that links to GTSAM—should enable -march=native. Check carefully, or you’ll hit runtime errors.

closed 2025-07-30 by ProfFan