borglab/gtsam #2255: Consider using sccache for (some) builds

Gold856, 2025-09-18:
## Feature

[sccache](https://github.com/mozilla/sccache/) is like ccache, but with remote storage as a built-in feature, including being able to store caches in the GitHub Actions cache. It can cache individual objects remotely and fetches them when needed.

## Motivation

Windows builds take an hour. :( And the fact that there are a limited number of runners means PR builds take much more than an hour for every push.

## Pitch

Consider using sccache to cache at least the Windows Release builds. Considering the sheer size of the built libraries, chances are that only one build can be cached without exceeding the GitHub Action cache limit (10GB), and caching Windows Release builds will probably save the most amount of time for the storage used.

## Alternatives

Not caching anything. Nothing is cached now, but we still get by.

Use an external cloud storage provider to store caches. sccache supports [many different storage options](https://github.com/mozilla/sccache/?tab=readme-ov-file#storage-options), but that generally costs money and requires authentication so people don't destroy your cache.

dellaert, 2025-09-30:
And Mac CI installs 2.2.3
```
Collecting numpy>=1.11.0 (from -r /Users/runner/work/gtsam/gtsam/python/requirements.txt (line 1))
  Downloading numpy-2.3.3-cp313-cp313-macosx_10_13_x86_64.whl.metadata (62 kB)
```

dellaert, 2025-09-30:
Oops, I'v been responding to wrong issue :-)