borglab/gtsam #2264: "pip install gtsam" doesn't enforce working combination of requirements for python bindings

timbarfoot, 2025-09-30:
## Description

It seems that a generic "pip install gtsam" does not install compatible requirements.  In particular, "numpy" gets installed and by default is a new version (e.g., 2.0.0) that seems to not be compatible with the C++ binary that the python bindings are linking to.  Downgrading "numpy" (e.g., to 1.26.4) seems to fix the issue, but this is not obvious and should either be documents or specified in "requirements.txt" so a compatible version gets automatically installed.

## Steps to reproduce

1. pip install gtsam   (in a python 3.9.6 venv - this automatically installed numpy 2.0.0 for me)
2. run any example - fails in a segmentation fault
3. downgrade numpy to 1.26.4 and the examples work as expected 

## Expected behavior

Examples should run properly without having to figure out that the wrong numpy gets installed.

## Environment

python 3.9.6 in venv

## Additional information

dellaert, 2025-09-30:
Thanks! I'll take a look.

dellaert, 2025-09-30:
FWIW, I’m on python 3.12 and numpy 2.2.6, and no issues :-/

dellaert, 2025-09-30:
Tim responded on slack: "but you’re probably building from scratch - for my use case I need an easier install so just using “pip install gtsam” - I did try several different versions of python (3.9, 3.10, 3.11, 3.12, 3.13) and they all seg fault…I only tested my working solution to downgrade numpy on 3.9"

timbarfoot, 2025-09-30:
yeah, just to be clear...the issue is getting the python bindings to link to the pre-built binary that gets installed with "pip install gtsam"

redheli, 2025-10-02:
i cannot reproduce the segmentation fault with `develop` branch
but do reproduce the issue in version `4.2` and have a fix

in short: old gtsam's pybind11(version 2.10) too old, can not handle numpy >2.0

i am not familar with gtsam pkg release, maybe someone has proper solution to apply the patch to fix old version

## how to reproduce (build from source with debuginfo)
```
$ git checkout 4.2
$ conda create -n gtsam_issue python=3.9.6
$ conda activate gtsam_issue
$ PYTHON_INCLUDE_DIR=$(python -c "from sysconfig import get_paths; print(get_paths()['include'])")
$ PYTHON_LIBRARY=$(find $CONDA_PREFIX/lib -name "libpython3.9*.so" | head -1)  # Picks libpython3.9.so or libpython3.9m.so
$ CONDA_ROOT=$CONDA_PREFIX
$ mkdir build && cd build
$ cmake -DCMAKE_BUILD_TYPE=Debug       -DGTSAM_BUILD_PYTHON=ON       -DGTSAM_PYTHON_STUBS=OFF       -DPython3_EXECUTABLE=$(which python)       -DPYBIND11_PYTHON_VERSION=3.9       -DPython3_INCLUDE_DIR=$PYTHON_INCLUDE_DIR       -DPython3_LIBRARY=$PYTHON_LIBRARY       -DPython3_FIND_STRATEGY=LOCATION       -DPython3_ROOT_DIR=$CONDA_ROOT       -DCMAKE_CXX_FLAGS="-Wno-deprecated-declarations -Wno-deprecated-copy" ..
$ make -j8
$ make python-install
```
run code `test_gtsam.py` :
```
import gtsam
import numpy as np

# Simple GTSAM example
graph = gtsam.NonlinearFactorGraph()
model = gtsam.noiseModel.Diagonal.Sigmas(np.array([1.0, 1.0, 1.0]))
```

$ python test_gtsam.py 
Segmentation fault (core dumped)

### trouble-shooting
gdb coredump shows crash happend when call nump api `PyArray_CopyInto`
```
int result = detail::npy_api::get().PyArray_CopyInto_(ref.ptr(), buf.ptr());
```

[image]

### how to fix
use pybind11 version 3.0.1 to replace wrap/pybind11, which is version `2.10`
just copy and replace folder wrap/pybind11

after fix, the script run without problem:
```
(gtsam_issue) max@max-laptop:scripts$ python test_gtsam.py 
GTSAM setup complete
```

timbarfoot, 2025-10-02:
yes, when I did "pip install gtsam" it installed 4.2 - again, to be clear, I'm not talking about building from source.  I'm talking about just trying to pip install then run the examples without any other extra work.

dellaert, 2025-10-02:
Yeah, `pip install gtsam` should work. Thanks both! We'll try the patch above for a minor 4.2 release.

suveergarg, 2025-10-07:
Maybe we can update develop with an updated version of pybind:  https://github.com/borglab/gtsam/pull/2276
Then we can create a new 4.2.1 tag or similar for the patch and run the pipeline to publish the package. I don't think I have the permission to create tags etc, so someone else can use the PR above and target it to the right branch and create a patched release.

ProfFan, 2025-10-08:
We need to do it from `wrap` instead, I can prepare a quick PR that does it.

timbarfoot, 2025-11-30:
@varunagrawal I'm not sure that I follow your logic to make numpy>=2.0.0.  This seems incompatible with the issue I raised.  I observed seg faults unless I specifically use numpy<2.0.0.  See my initial issue message.  As I said there, numpy==1.26.4 works fine but then it's 2.0.0 and beyond and it fails at runtime.  

Or are you saying that the upgrade to numpy>=2.0.0 only works for `pip install gtsam-devel' and not 'pip install gtsam'?